OpenIPC/ipctool #14: Incorrect sensor detection on two test cameras

widgetii, 2021-11-25:
```
172.17.32.51# ./ipc_chip_info 
Chip: HiSilicon 3516cv300
Sensor: Sony IMX323

172.17.32.52# ./ipc_chip_info 
Chip: HiSilicon 3516cv300
Sensor: Sony IMX323
```
`cat /proc/umap/sys` on both shows `ar0237` type but `Sofia` detects as `SC2235`:

```
[DEBUG]: mirror=0
[DEBUG]: Flip=0
[DEBUG]: ~~~~~~~~~~SC2235_init  JJ ~~~~~~~~~
[DEBUG]: Denoise level:  Day[3]      Night[3]
[DEBUG]: Denoise level:  Day[3]      Night[3]
[DEBUG]: IR: 65
```

widgetii, 2022-01-28:
Checked once again (this time both cameras are on OpenIPC firmware) and it's confirmed that the real sensor name is `IMX323`:

```
sensors:
- vendor: Sony
  model: IMX323
  control:
    bus: 0
    type: i2c
    addr: 0x34
  data:
    type: DC
  clock: 27MHz
  ```

closed 2022-01-28 by widgetii

minikdo, 2022-01-29:
I have the same issue:

`# cat /proc/umap/sys

[SYS] Version: [Hi3516CV300_MPP_V1.0.4.0 B050 Release], Build Time[Jun  3 2018, 21:42:04]

-----MODULE PARAM---------------------------------------------------------------
vi_vpss_online  en_online_delayint  sensor0   sensor1  mem_total
             0                   0ar0237_i2c_dc        NA        64`

`# ipcinfo --long-sensor
imx323_i2c`