OpenIPC/ipctool #18: Collecting information about SigmaStar devices

ZigFisher, 2021-11-25:
Recently I found this output of information. 
Perhaps it will be useful when researching the original firmware.

```
root@uniview-dev:~# cat /sys/devices/soc0/machine

INFINITY6B0 SSC009A-S01A QFN88

```

p0i5k, 2022-09-25:
CamHi probably SSC338Q
```
cat /sys/devices/soc0/machine
INFINITY6E SSC012B-S01A

cat /sys/devices/soc0/soc_id
241

cat /sys/devices/soc0/family
I6E

cat /sys/devices/soc0/revision
2

cat /sys/class/mstar/msys/CHIP_ID
Chip_ID: 0xF1

cat /sys/class/mstar/msys/CHIP_VERSION
Chip_Version: 1

cat /sys/class/mstar/msys/TEMP_R
Temperature 65

cat /sys/devices/system/miu/miu0/dram_info
DRAM Type:   DDR3
DRAM Size:   256MB
DRAM Freq:   1809MHz
MIUPLL Freq: 400MHz
Data Rate:   4x Mode
Bus Width:   16bit
SSC:         OFF

```

dimerr, 2023-09-15:
SSC377
```
cat /sys/devices/soc0/machine
INFINITY6C SSC027A-S01A
cat /sys/devices/soc0/soc_id
249
cat /sys/devices/soc0/family
I6C
cat /sys/devices/soc0/revision
1
cat /sys/class/mstar/msys/CHIP_ID
Chip_ID: 0xF9
cat /sys/class/mstar/msys/CHIP_VERSION
Chip_Version: 0
cat /sys/class/mstar/msys/TEMP_R
Temperature 64
cat /sys/devices/system/miu/miu_bw0/dram_info 
DRAM Type:   Unknown
DRAM Size:   1MB
DRAM Freq:   1332MHz
MIUPLL Freq: 200MHz
Data Rate:   4x Mode
Bus Width:   16bit
SSC:         OFF
```

gitgayhub, 2023-09-18:
SSC339G
from a Dahua camera, no shell access, information from bootlog

```
OF: fdt:Machine model: INFINITY6E SSC013A-S01A
Version : MVX4##I6E#g#######KL_LX409##[BR:g]#XVM
```
```
IPL 50a09f3
D-b7
MCP32_2133_4X 6L
miupll_400MHz
1024MB
BIST0_0001-OK
SW ReadDQS Training
SW WriteDQS Training
HS on
Load IPL_CUST from SPINAND

 BlSize 00006330
CUST_KEY
Checksum OK
JMP+++
IPL_CUST 50a09f3
_IPLCustSNandOp
Load BL from SPINAND
CUST Key
KEYN_SIZE(0x0100)
KEYN_ADDRESS(0x23c06130)
  decomp_size=0x00041e00
```