OpenIPC/ipctool #30: tftp as alternative method and new T10 camera

brainstorm, 2022-08-15:
My MIPS-based camera has a `tftp` binary and no NFS support on mount, therefore:

## On a desktop machine

```
$ pip install ptftpd
$ ptftpd -p 6969 en0 /Users/rvalls/Downloads
```

## On the embedded device (camera)

```
# tftp -r /Users/rvalls/Downloads/ipctool-mips32 -g 192.168.1.107 6969

/Users/rvalls/Downlo  46% |**************                 | 61952   0:00:01 ETA
/Users/rvalls/Downlo  97% |****************************** |   126k  0:00:00 ETA
/Users/rvalls/Downlo 100% |*******************************|   129k  0:00:00 ETA
/Users/rvalls/Downlo 100% |*******************************|   129k  0:00:00 ETA
```

Then the resulting YAML file [for this camera](https://www.gbconline.it/en/prodotti/schedaprodotto?codice=67.6894.03) with [this bootlog](https://pastebin.com/DD4S0uQd) looks like this:

```
/tmp # ./ipctool-mips32 
---
chip:
  vendor: Ingenic
  model: T10
ethernet:
  mac: "40:ffaa:56:32:13:ff"
rom:
  - type: nor
    block: 32K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 3ed97345
      - name: kernel
        size: 0x190000
        sha1: 5e503445
      - name: root
        size: 0x300000
        sha1: c67bd327
      - name: appfs
        size: 0x330000
        path: /system,jffs2,rw
    size: 8M
ram:
  total: 64M
  media: 0M
firmware:
  kernel: "3.10.14 (PREEMPT Tue Oct 30 18:55:41 PDT 2018)"
  toolchain: gcc version 4.7.2 (Ingenic r2.3.3 2016.12) 
  libc: uClibc 0.9.33.2
```

It didn't really collect any information about the sensor, UBoot version, etc... like the example on the README.md though :/

Thanks @ZigFisher for your assistance over Gitter!

brainstorm, 2022-08-15:
Here are some further hints that you might want to use to improve this tool?:

```
/system/sensor # ls
jxh62.bin        sensor_jxh62.ko

/system/init # lsmod
audio 72000 0 - Live 0xc0442000 (O)
sensor_jxh62 8466 0 - Live 0xc0421000 (O)
tx_isp 361277 0 - Live 0xc03b1000 (O)
mtnet7601Usta 28130 1 - Live 0xc0264000 (O)
mt7601Usta 760508 1 mtnet7601Usta, Live 0xc0188000 (O)
mtutil7601Usta 47826 2 mtnet7601Usta,mt7601Usta, Live 0xc0084000 (O)
```

widgetii, 2022-08-15:
PRs are always welcome!

brainstorm, 2022-09-15:
> PRs are always welcome!

Sure thing! Done in #39

closed 2022-09-15 by brainstorm