OpenIPC/ipctool #36: Run ipctool from SD card

NathanBxer, 2022-09-04:
Hi, any chance of some instructions on how to run ipctool from an sdcard or from a serial connection?

ZigFisher, 2022-09-04:
Hi, write the type of device, model, manufacturer and attach a photo.
The question is too voluminous to give an answer without an explanation on your part.

NathanBxer, 2022-09-04:
I have 3 kinds first one is my cheap test one, i kind of got openipc on it. Ill make 3 posts to cover them,

[image]

[image]

[image]

[image]

```
System startup
U-Boot 2010.06 (Feb 27 2017 - 21:45:11)
Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xef 0x40 0x17
spi_general_qe_enable(294): Error: Disable Quad failed! reg: 0x2
Block:64KB Chip:8MB Name:"W25Q64FV"
SPI Nor total size: 8MB
MMC:   
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
*** Warning - bad CRC, using default environment
In:    serial
Out:   serial
Err:   serial
Hisilicon ETH net controler
hieth_init :0
MAC:   00-00-23-34-45-66
PHY not link.
Hit any key to stop autoboot:  2  1  0 
8192 KiB hi_fmc at 0:0 is now current device
## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2440984 Bytes = 2.3 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK
Starting kernel ...
Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0
Linux version 3.4.35 (zg@ESNTD_ZGLUX) (gcc version 4.8.3 20131202 (prerelease) (Hisilicon_v300) ) #41 Sat Nov 19 17:04:33 CST 2016
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: hi3518ev200
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 root=/dev/mtdblock4 rootfstype=squashfs mtdparts=hi_sfc:256k(boot),128k(env),128k(conf),2560k(os),5120k(rootfs)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 32MB = 32MB total
Memory: 27476k/27476k available, 5292k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc044a000   (4360 kB)
      .init : 0xc044a000 - 0xc046a464   ( 130 kB)
      .data : 0xc046c000 - 0xc049a140   ( 185 kB)
       .bss : 0xc049a164 - 0xc04d8bb8   ( 251 kB)
SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:32
VIC @fe0d0000: id 0x00641190, vendor 0x41
sched_clock: 32 bits at 49MHz, resolution 20ns, wraps every 86767ms
Console: colour dummy device 80x30
[... 1181 lines cut ...]
     --      Y        32        32        32

-----VI WDR ATTR---------------------------------------------------------------
    Mode  BufNum    DstW    DstH  PoolId   VcNum  DesNum   State  bCompress
    NONE       0    1280     720      -1       0       0    Init      N

-----VI WDR DES STATUS----------------------------------------------------------
     Idx    IntGap    IntCnt  CcErrCnt

-----VI WDR SRC STATUS----------------------------------------------------------
     Idx    IntGap    IntCnt  CcErrCnt

-----VI WDR COMBINE STATUS--------------------------------------------------------
    IntGap    IntCnt  CcErrCnt
         0         0         0
root@openipc-hi3518ev200:~# 
root@openipc-hi3518ev200:~# 
root@openipc-hi3518ev200:~# cat /proc/umap/vi[J[J[J
cat: read error: Is a directory
root@openipc-hi3518ev200:~# 
root@openipc-hi3518ev200:~# cat /proc/umap/[J[J
cat: read error: Is a directory
root@openipc-hi3518ev200:~# 
root@openipc-hi3518ev200:~# cat /proc/umap[J[J[J[J[J[J[J[J[J[J[J[J
```

NathanBxer, 2022-09-04:
Next one is solar cam

```

![20220730_224655](https://user-images.githubusercontent.com/84220506/188312256-35ac4e76-0bf0-4d4b-9c99-ebec7ee27ea3.jpg)
![20220731_002935](https://user-images.githubusercontent.com/84220506/188312264-a55c9159-3884-464c-87ae-b1dd861bf4c9.jpg)
![20220815_213557](https://user-images.githubusercontent.com/84220506/188312274-8039c938-6b86-4464-8fec-a855a59b8865.jpg)
![20220815_213628](https://user-images.githubusercontent.com/84220506/188312278-69b899e5-800c-4327-9d34-eca8aa8d4380.jpg)
![20220815_213714](https://user-images.githubusercontent.com/84220506/188312280-e0f87578-f344-4773-9b2e-b80645f77abb.jpg)
![20220815_213739](https://user-images.githubusercontent.com/84220506/188312287-2acc20e3-5a58-48e0-8cf3-401af931c4c2.jpg)
![20220815_214308](https://user-images.githubusercontent.com/84220506/188312292-9cfb6034-93ea-4d69-a691-dbd86c697af0.jpg)
![20220815_214416](https://user-images.githubusercontent.com/84220506/188312298-658ab9d4-9ffe-4a8c-9741-920650dd30af.jpg)
![20220815_215004](https://user-images.githubusercontent.com/84220506/188312300-7643ac94-b4b9-4ee0-a510-f96b1294ae9f.jpg)
![20220815_215108](https://user-images.githubusercontent.com/84220506/188312308-eba9eb8f-c480-401e-91f7-8b972e0c5bf6.jpg)

uboot for this one

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.09.04 14:02:44 =~=~=~=~=~=~=~=~=~=~=~=
                         
System startup
U-Boot 2010.06 (Sep 14 2018 - 18:47:03)
Check Flash Memory Controller v100 ... Found
SPI Nor(cs 0) ID: 0xef 0x60 0x17
spi_w25q256fv_qe_enable(193): Error: Disable Quad failed! reg:0x22
Block:64KB Chip:8MB Name:"W25Q64FW"
SPI Nor total size: 8MB
MMC:   
EMMC/MMC/SD controller initialization.
Card did not respond to voltage select!
No EMMC/MMC/SD device found !
In:    serial
Out:   serial
Err:   serial
uboot-other-mipi-init
8192 KiB hi_fmc at 0:0 is now current device
uboot mirror:07 flip:04
uboot_get_init_exposure ==<adc=180 i=13 aec_h:0, aec_l:2 agc_h:9 agc_l:123>
==========================================
=== PS5230 Sensor UBoot init success!=====
Press Ctrl+C to stop autoboot
## Starting application at 0x80008000 ...
********hello Huawei LiteOS ARM926********
version : Huawei LiteOS V200R001C10B031
open-version : Huawei LiteOS 1.4.6
build data : Aug 21 2019 18:44:38
******************
Huawei LiteOS # Spi Nor ID:0xEF 0x60 0x17 0x00 0x00 0x00 0x00 0x00 
Spi Nor Flash Info:
Name:"W25Q64FW" Size:8MB Block:64KB
Calling lwIPRegSecSspCbk
********mmc host init ! ********
Date:Oct 14 2019. Time:21:46:58.
osal_proc_mkdir - parent is NULL! proc=0x8093e99c 
[       38]ubia_wdg_feed_thread===>>
mem_start=0x80000000, MEM_OS_SIZE=33M, MEM_USB_SIZE=1M, mmz_start=0x82200000, mmz_size=30M
mmz param= anonymous,0,0x82200000,30M
<6>Hisilicon Media Memory Zone Manager
load sys.ko for Hi3518EV200...OK!
load tde.ko ...OK!
load region.ko for Hi3518EV200...OK!
load vgs.ko for Hi3518EV200...OK!
load viu.ko for Hi3518EV200...OK!
ISP Mod init!
load vpss.ko for Hi3518EV200...OK!
Load hifb.ko OK!
load rc.ko ...OK!
load venc.ko ...OK!
load chnl.ko for Hi3518EV200...OK!
load h264e.ko ...OK!
load jpege.ko ...OK!
[... 370 lines cut ...]
[    34498][ULOG_INFO]rmmAudioPlayTh[1382] option:5654
[    34498][ULOG_INFO]rmmAudioPlayTh[1576] rmmAudioPlayTh===>READY_FOR_PAIR[34498]
[    34506][ULOG_INFO]wav_open[70] u32RiffSize:0x145b6
[    34510][ULOG_INFO]wav_open[93] u32FmtSize:0x10!
[    34515][ULOG_INFO]wav_open[107] u32SamplesPerSec:16000
[    34521][ULOG_INFO]wav_open[113] data.size:0x14570!
[    34526][ULOG_INFO]rmmAudioPlayTh[1793] rmmAudioPlayTh ----> fd[2154665312] start play[duration:0 prio:1]
[    34536]audio_switch_on===>>
[    36305][ULOG_INFO]recognizerStart[209] ------------------recognize start
[    36655][ULOG_INFO]ubia_read_wav_data[822] End of file!
[    36655][ULOG_INFO]playSpecialAudio[1118] End of file,frameNo=0
[    36661][ULOG_INFO]playSpecialAudio[1158] ====<Finish Count:0
[    37580]audio_switch_off===>>
[    37580][ULOG_INFO]rmmAudioInTh[674] [37] playing end
[    37718][ULOG_INFO]recognizerEnd[281] ------------------recognize invalid data, errorCode:100, error:not enough signal
[    37723][ULOG_INFO]recognizerEnd[283] string:(null)
[      118][ULOG_INFO]ubia_rtc_get[117] 1970-01-01 0:3:4   utsec:184

[      209]hisi_mpp_init===>>
[      217][ULOG_INFO]hisi_vid

```

Camera instructions
[Solar WFi CCTV Camera.pdf](https://github.com/OpenIPC/ipctool/files/9484624/Solar.WFi.CCTV.Camera.pdf)

NathanBxer, 2022-09-04:
last one is swann-806cam

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

```

U-Boot 2010.06-svn (May 12 2015 - 16:55:46)

DRAM:  256 MiB
Check spi flash controller v350... Found
Spi(cs1) ID: 0xC2 0x20 0x17 0xC2 0x20 0x17
Spi(cs1): Block:64KB Chip:8MB Name:"MX25L6406E"
envcrc 0x5569e8fe
ENV_SIZE = 0xfffc
In:    serial
Out:   serial
Err:   serial
Press Ctrl+C to stop autoboot
hisilicon # printenv
bootcmd=setenv setargs setenv bootargs ${bootargs};run setargs;fload;bootm 0x82000000
baudrate=115200
bootfile="uImage"
da=mw.b 0x82000000 ff 1000000;tftp 0x82000000 u-boot.bin.img;sf probe 0;flwrite
du=mw.b 0x82000000 ff 1000000;tftp 0x82000000 user-x.cramfs.img;sf probe 0;flwrite
dr=mw.b 0x82000000 ff 1000000;tftp 0x82000000 romfs-x.cramfs.img;sf probe 0;flwrite
dw=mw.b 0x82000000 ff 1000000;tftp 0x82000000 web-x.cramfs.img;sf probe 0;flwrite
dc=mw.b 0x82000000 ff 1000000;tftp 0x82000000 custom-x.cramfs.img;sf probe 0;flwrite
up=mw.b 0x82000000 ff 1000000;tftp 0x82000000 update.img;sf probe 0;flwrite
ua=mw.b 0x82000000 ff 1000000;tftp 0x82000000 upall_verify.img;sf probe 0;flwrite
tk=mw.b 0x82000000 ff 1000000;tftp 0x82000000 uImage; bootm 0x82000000
dd=mw.b 0x82000000 ff 1000000;tftp 0x82000000 mtd-x.jffs2.img;sf probe 0;flwrite
netmask=255.255.255.0
ethaddr=00:12:13:16:2d:18
HWID=8043420004748432
NID=0x0012
muxctl0=
muxval0=
gpio0=
gpioval0=
osmem=39M
appVideoStandard=PAL
ipaddr=192.168.1.13
appSystemLanguage=French
bootdelay=5
serverip=192.168.1.19
bootargs=mem=39M console=ttyAMA0,115200n8 root=/dev/mtdblock1 rootfstype=cramfs mtdparts=hi_sfc:256K(boot),3520K(romfs),2560K(user),1280K(web),256K(cus),320K(mtd)
myboot=setenv bootargs mem=40M console=ttyAMA0,115200n8 panic=10 ; tftp 0x82000000 cam/uImage ; bootm 0x82000000
stdin=serial
stdout=serial
stderr=serial
verify=n
ver=U-Boot 2010.06-svn (May 12 2015 - 16:55:46)

Environment size: 1444/65532 bytes

```

closed 2026-09-16 by openipc-ai