OpenIPC/ipctool #37: Upgrade / Restore fails on Videopark devices

DSchndr, 2022-09-14:
Hi, I've tested upgrade / restore functunality of the tool and it fails on my devices
I have some devices to test and can flash ofw back, how should I debug the flash process?

Also, should i upload a dump somewhere? I did not see the hunter streamer binary anywhere

```
/tmp # ./ipctool
---
chip:
  vendor: HiSilicon
  model: 3516CV300
ethernet:
  mac: "e0:61:b2:4e:a6:ff"
  u-mdio-phyaddr: 1
  phy-id: 0x001cc816
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 48195e9b
        contains:
          - name: uboot-env
            offset: 0x20000
      - name: defcfg
        size: 0x10000
        sha1: 67d9fc96
      - name: kernel
        size: 0x1f0000
        sha1: 7625fbaa
      - name: root
        size: 0x960000
        path: /,squashfs
        sha1: cb44f590
      - name: custom
        size: 0x400000
        sha1: f427120a
      - name: config
        size: 0x60000
        sha1: 928813b9
    size: 16M
    addr-mode: 3-byte
ram:
  total: 256M
  media: 144M
firmware:
  kernel: "3.18.20 (Wed Apr 4 14:46:18 CST 2018)"
  toolchain: gcc version 4.9.4 20150629 (prerelease) (Hisilicon_v500_20150831)
  libc: uClibc 0.9.33.2
  sdk: "Hi3516CV300_MPP_V1.0.1.0 B070 Release (Feb 13 2017, 20:42:49)"
  main-app: hunter
sensors:
- vendor: Sony
  model: IMX323
  control:
    bus: 0
    type: i2c
    addr: 0x34
  data:
    type: DC
  clock: 37.125MHz

```
```
~ # cat /proc/partitions
major minor  #blocks  name

  31        0        256 mtdblock0
  31        1         64 mtdblock1
  31        2       1984 mtdblock2
  31        3       9600 mtdblock3
  31        4       4096 mtdblock4
  31        5        384 mtdblock5
~ # cat /proc/mounts
rootfs / rootfs rw 0 0
/dev/root / squashfs ro,relatime 0 0
devtmpfs /dev devtmpfs rw,relatime,size=53760k,nr_inodes=13440,mode=755 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
tmpfs /tmp tmpfs rw,relatime 0 0
tmpfs /var tmpfs rw,relatime 0 0
devpts /dev/pts devpts rw,relatime,mode=600 0 0
/dev/mtdblock4 /mnt/special squashfs ro,relatime 0 0
/dev/mtdblock5 /mnt/mtd jffs2 rw,sync,relatime 0 0

```
```
~ # dmesg | grep hisi_spi_nor.0
hisi-sfc hisi_spi_nor.0: all blocks is unlocked.
hisi-sfc hisi_spi_nor.0: gd25q128 (16384 Kbytes)

```
```
/tmp # ./ipctool upgrade upgrade.cv300
There was no Sofia process detected
Use --force switch to skip the check
aborting...
/tmp # ./ipctool upgrade upgrade.cv300 --force
There was no Sofia process detected
Unmounting /mnt/special
Unmounting /mnt/mtd
Using 'upgrade.cv300' as upgrade bundle
        boot    0xb68d4010, size: 262144 bytes
        env     0x63010, size: 65536 bytes
        kernel  0xb66f3010, size: 1966080 bytes
        rootfs  0xb6272010, size: 4718592 bytes
Analyzing boot
Analyzing env
Analyzing kernel
Analyzing rootfs
Upgrading boot
Flashing [wwww]
Upgrading env
Flashing [w]
Upgrading kernel
Flashing [wwwwwwwwwwwwwwwwwwwwwwwwwwwwww]
Upgrading rootfs
Could not open mtd device: 46.....................................................]

Something went wrong, aborting...
Early exit, check the logs

```
```
/tmp # ./ipctool restore bak
Restoring the backup

Loading backup from file bak...
Are you sure to proceed? (y/n)? y
Checking boot...
[0] 0x00000000  0x00040000      48195e9b        48195e9b
Checking defcfg...
[1] 0x00040000  0x00010000      67d9fc96        67d9fc96
Checking kernel...
[2] 0x00050000  0x001f0000      7625fbaa        7625fbaa
Checking root...
[3] 0x00240000  0x00960000      cb44f590        cb44f590
Checking custom...
[4] 0x00ba0000  0x00400000      f427120a        f427120a
Not checked (no SHA1)
[5] 0x00fa0000  0x00060000           n/a
Backups were checked
Analyzing boot
Analyzing defcfg
Analyzing kernel
Analyzing root
Analyzing custom
Analyzing config
Restoring boot
Flashing [wwww]
Restoring defcfg
Flashing [w]
Restoring kernel
Fail to erase +0x40000...................]

Something went wrong, aborting...
```

widgetii, 2022-09-14:
The best way from my point of view is to add more debug output to understand what's going on while `Could not open mtd device` phase. This code was tested and works only in couple XM boards, so it definitely needed to be generalized

DSchndr, 2022-09-14:
Hmm... flashrom gives me "MTD size is not a power of 2" on those problematic partitions
Might be related?

widgetii, 2022-09-15:
For SPI NOR flashes there are two sequential operations to write new data:
1) Erase block of data (to make all bits in it have 1)
```
root@openipc-hi3516ev300:~# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00040000 00010000 "boot"
mtd1: 001f0000 00010000 "kernel"
mtd2: 00500000 00010000 "rootfs"
mtd3: 008d0000 00010000 "rootfs_data"
```
Block size is depends on IC flash type but nowadays, 0x10000 size is commonly used

2) Write actual data (this operation could be done against even single byte but only one time before new erase), basically "erased" bit set to 1 could be changed to 0 (or is not touched at all)

DSchndr, 2022-09-15:
I dumped the flashrom idea, flash is not accessible over spidev and it cannot map mtd partitions (+ the ^2 error) :/

Just noticed something, erasing fails at 0x40000, exactly where defcfg romfs starts (or well, mtd1)
One of the first things the init script does is to `insmod /usr/bin/modules/devcfg.ko`, could it be that this module interferes if it reserves some memory from the partition?

At a quick glance it handles "super secure" encryption/hwid crap and product information, the data that is probably in romfs (since it is not in my binwalk extract)

widgetii, 2022-09-15:
Your flash is also may be locked by the hardware manufacturer (some of them supports locks for specific areas)

DSchndr, 2022-09-15:
Argh, could check on the bricked one with asprogrammer if sreg bits are set but I think they are not
Well, might as well use the oppertunity to flash openipc xD
Bootlog says that flash is unlocked too

widgetii, 2022-09-15:
Flash may be also "locked" in vendor Linux kernel by forbidding specific areas to erase/write

DSchndr, 2022-09-15:
Tought about that too since uart is completely locked down but that module can also write apparently (does not mean that it will)
Have to test some ideas, will report later

Btw, has someone reverse engineered hisi-sfc / the flash interface? Direct access is best - I think - bypassing possible locks

widgetii, 2022-09-15:
It's well documented in the [official datasheets](https://github.com/openipc/camerasrnd#chip-families-information) and all Linux drivers are open too

DSchndr, 2022-09-15:
Ok, did a simple `echo "test" > /dev/mtd1` and it did set some bits in the first 4 bytes to 0 at 0x400000
So writing works
Sreg is also normal: 00000000(0x00), 00000000(0x00), 11000000(0xC0)

I wonder where the 46 in "Could not open mtd device: 46" comes from, seems like a miscalculation since partition 46 does not exist for sure (if i understand the code correctly)

How should I build this tool? To debug this issue further

widgetii, 2022-09-15:
Download toolchain for your camera (for cv300 there is [a link](https://github.com/OpenIPC/firmware/releases/download/latest/arm926t-gcc8.4.0-musl-3_18-75d977ad.tgz) ), unpack somewhere, add `bin` directory to `PATH` and ensure that `arm-openipc-linux-musleabi-gcc` is running

Create toolchain file named `arm-openipc-linux-musleabi.cmake`
```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_CROSSCOMPILING TRUE)

# Specify the cross compiler.
set(CMAKE_C_COMPILER arm-openipc-linux-musleabi-gcc CACHE FILEPATH "C compiler")
set(CMAKE_CXX_COMPILER arm-openipc-linux-musleabi-g++ CACHE FILEPATH "C++ compiler")
set(CMAKE_ASM_COMPILER arm-openipc-linux-musleabi-as CACHE FILEPATH "ASM compiler")

# Search libraries only under *target* paths.
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
```

Then configure and build via CMake:
```
cmake -H. -Bbuild -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=arm-openipc-linux-musleabi.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
cmake --build build
```

`build/ipctool` will be a target binary

DSchndr, 2022-09-15:
🎉 
[image]

Will add code for feeding watchdog and killing applications later
I did it like the updater from the streamer: send **sigtrap** to hunter & sysdaemon (it does more but thats the working bare minimum) then start the update

Seems like just killing hunter and leaving sysdaemon running caused this? no motivation to test, I can confirm that I had 3 bricks while testing xD

By the way, is there a way to make a pack which has the (16m spi) memory type preconfigured? not having to take the device apart would be the cherry on top

Since I have multiple devices and they are absolutely not documented at all: how about a device repository where pinouts, pictures, dumps, installation guides and configs can be shared?

Anyways, thanks for support and the projects (still a wip but already much, much better than the stock spyware :) )

widgetii, 2022-09-15:
Congrats from my side

`By the way, is there a way to make a pack which has the (16m spi) memory type preconfigured? `

Everything is possible through the code (just make it more universal and suitable for work with many vendors at the same time: for example, it now kills Sofia main process, but it should detect environment by itself and make appropriate steps for specific vendor)

DSchndr, 2022-09-16:
Hah, I did it! :D
~5 telnet commands for stock -> openipc on these cameras
ipctool flashes uboot and reboots, uboot checks firstboot env; if it is it flashes images from tftp, sets 16m mtd and boots into openipc ultimate

why this? rootfs gets mangled + the binary with hidden tftp is extremely bad so I let uboot handle it - it is fast enough, the  watchdog can be ignored too :)

Process takes only a couple of minutes and can be completely automated (since telnet is enabled on boot till hunter is started)

closed 2022-09-16 by DSchndr