OpenIPC/ipctool #44: run ipctool on Goke GK7205V300

JeromeDuboisVizzia, 2022-11-04:
Hi all,
First, thanks for the wonderful work done on ipctool.
I have a vanilla Goke GK7205V300 SoC with http admin access and would like to install ipctool.
From the ipctool installation instructions, it seems I need to have some kind of terminal connection to the SoC.
I did a nmap, and port 23 is open, so I should be able to telnet and install ipctool, but when asked for login/password, I don't know what to put.

Is there any other way to install ipctool, or do you know how we can telnet to this SoC?
I do have admin access through http.

Thanks in advance for your help.
Best,
Jérôme.

FYI nmap results:
```bash
nmap -v -sV -sT -p- -oA nmap_full 192.168.8.130
Starting Nmap 7.80 ( https://nmap.org/ ) at 2022-11-04 12:47 CET
NSE: Loaded 45 scripts for scanning.
Initiating ARP Ping Scan at 12:47
Scanning 192.168.8.130 [1 port]
Completed ARP Ping Scan at 12:47, 0.06s elapsed (1 total hosts)
Initiating Parallel DNS resolution of 1 host. at 12:47
Completed Parallel DNS resolution of 1 host. at 12:47, 0.00s elapsed
Initiating Connect Scan at 12:47
Scanning e0:3c:5b:95:ac:06 (192.168.8.130) [65535 ports]
Discovered open port 23/tcp on 192.168.8.130
Discovered open port 80/tcp on 192.168.8.130
Discovered open port 8080/tcp on 192.168.8.130
Discovered open port 554/tcp on 192.168.8.130
Discovered open port 3210/tcp on 192.168.8.130
Discovered open port 8554/tcp on 192.168.8.130
Discovered open port 13452/tcp on 192.168.8.130
Discovered open port 1935/tcp on 192.168.8.130
Completed Connect Scan at 12:47, 9.53s elapsed (65535 total ports)
Initiating Service scan at 12:47
Scanning 8 services on e0:3c:5b:95:ac:06 (192.168.8.130)
```

widgetii, 2022-11-04:
What board do you have? Please make a photo

JeromeDuboisVizzia, 2022-11-04:
Thanks for your reply!
At the moment, the board is hardly visible as there is a large radiator installed on the SoC.
The whole thing comes from https://www.aliexpress.com/item/1005004537345733.html .
From this link you can view:
1) the whole package (boards + camera)
2) in the "excel sheet" a view of the board + technical information

Hope it helps :-/.
Best,
J.

widgetii, 2022-11-04:
You can try to override bootcmd in U-Boot and add `init=/bin/sh` to get console access, then launching `/etc/init.d/*` script to restore normal board boot flow

BTW, there is no IMX385 public driver available for this SoC yet

JeromeDuboisVizzia, 2022-11-04:
Hmm, not sure if I understand completely, so sorry for the dumb question: How do we do the bootcmd override?
As of now, we connect to http://X.Y.Z.A with the admin account, to access the web admin UI where we can see images, and configure some options.
From this UI, I do not see any U-Boot options.
Maybe you are referring to some other way of connecting to the device?

widgetii, 2022-11-04:
Start from [these materials](https://github.com/openipc/camerasrnd#articlesbooks-for-newbies)

JeromeDuboisVizzia, 2022-11-04:
Thanks for your reply, we'll have a look at these.
We can close this issue as it might not be related to ipctool directly.
Great job, super responsiveness!

closed 2022-11-04 by JeromeDuboisVizzia