OpenIPC/ipctool #52: gc1024 is not detected

Moleus, 2023-01-15:
`ipctool -s` returns a message  `Error: unexpected value for SuperPix == 0x78`.

### Manual detection
I've followed suggestions of OpenIPC's telegram chat members and ran:
```
# ./ipctool i2cdetect
       0  1  2  3  4  5  6  7   8  9  a  b  c  d  e  f
     : xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |
                      ....
   70: xx xx xx xx xx xx xx xx  78 79 xx xx xx xx xx xx  |
                      ....
  f0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx
```
The sensor id can be found at `f0` - `10 24`. (Unnecessary lines are removed, file with the original output - [i2cdump_0x78.txt](https://github.com/OpenIPC/ipctool/files/10420829/i2cdump_0x78.txt))
```
# ./ipctool i2cdump 0x78 0 0x3ff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
    : 00 1F FF 03 98 01 6F 00  A6 00 00 00 08 02 E6 05  |  ......o.........
  10: 08 00 18 11 01 00 C0 17  0A 06 11 4F 11 10 F8 38  |  ...........O...8
...
  e0: 00 AB B6 80 9D EC F1 C7  DA 00 00 00 00 00 1D 40  |  ...............@
  f0: 10 24 0F 00 03 00 00 B9  03 0E 00 78 C4 11 00 00  |  .$.........x....
 100: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ...............
...
 1f0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
 200: 03 03 03 03 03 03 03 03  03 03 03 03 03 03 03 03  |  ................
 ...
 2d0: 03 03 03 03 03 03 03 03  03 03 03 03 03 03 03 03  |  ................
...
 3f0: 98 98 98 98 98 98 98 98  98 98 98 98 98 98 98     |  ...............
```
### The problem
According to the detection function source code, `3f0` or `3f1` should be `FF` to read the value from `f0` address. 

https://github.com/OpenIPC/ipctool/blob/11f89463b1341c8eaf86342d2940ac9134cdc108/src/sensors.c#L664-L675
But it's not the case, because `3f0` and `3f1` are filled with a garbage. You can see the output of multiple executions below:
```
# ./ipctool i2cdump 0x78 0x3f0 0x3ff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
 3f0: CE CE CE CE CE CE CE CE  CE CE CE CE CE CE CE
# ./ipctool i2cdump 0x78 0x3f0 0x3ff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
 3f0: 50 50 50 50 50 50 50 50  50 50 50 50 50 50 50
# ./ipctool i2cdump 0x78 0x3f0 0x3ff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
 3f0: 59 59 59 59 59 59 59 59  59 59 59 59 59 59 59
``` 
###  Side-effects  
Also, garbage in `fa` and in `fb` addresses prevents the early break in SuperPix detection function
https://github.com/OpenIPC/ipctool/blob/11f89463b1341c8eaf86342d2940ac9134cdc108/src/sensors.c#L746-L749
and execution falls into the default case
https://github.com/OpenIPC/ipctool/blob/11f89463b1341c8eaf86342d2940ac9134cdc108/src/sensors.c#L768-L769 
, so we get an error message `Error: unexpected value for SuperPix == 0x78`, but it's not a SuperPix sensor

### Solutions?
I'm knew to this topic and don't have a fundamental knowledge in sensors/i2c stuff, so I can't say if it's OK to have a "garbage" in a memory and how to deal with it. But can we check the value of `f0` address before `3f0`?

dimerr, 2023-01-15:
What's your soc?

Moleus, 2023-01-15:
hi3518ev200
```
# cat cpuinfo
Processor       : ARM926EJ-S rev 5 (v5l)
BogoMIPS        : 269.10
Features        : swp half thumb fastmult edsp java
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 5

Hardware        : hi3518ev200
Revision        : 0000
Serial          : 0000000000000000
```
The output of `ipctool` - [ipctool_output.txt](https://github.com/OpenIPC/ipctool/files/10420957/ipctool_output.txt)

I've looked at a [table](https://github.com/themactep/openipc-wiki/blob/master/en/hardware-sensor-identification.md) with sensor identifiers provided by Paul, and saw that `GC1024` has chip_id `0x1004`. That's strange.

I'm tinkering with `VstarCam C7824WIP` and found a note about it's `GC1024` sensor  - [code](https://github.com/fthiebolt/hi35xx-buildroot-c7824wip/blob/e10a041356e97b6e216fde02f6ddd04c2729a919/package/myopencam/tests/Makefile#L43-L48). Does this mean that it's the `GC1024` with invalid id?

dimerr, 2023-01-15:
try again

Moleus, 2023-01-15:
Got the ipctool with the patch. Still doesn't work. Same error. 

I think that this condition fails, because `0x78` not listed in `possible_i2c_addrs`
https://github.com/OpenIPC/ipctool/blob/34ef8b2fb2018883761fa88805081fdc7e9ce237/src/sensors.c#L791 

As far as I understand list of possible addresses for hisilicon is declared here:
https://github.com/OpenIPC/ipctool/blob/34ef8b2fb2018883761fa88805081fdc7e9ce237/src/hal/hisi/hal_hisi.c#L28-L35

What if we add `0x78` to this array?

dimerr, 2023-01-15:
it's already added

Moleus, 2023-01-15:
I see, in this https://github.com/OpenIPC/ipctool/commit/34ef8b2fb2018883761fa88805081fdc7e9ce237 commit you've only added the case statement, and might forgot to push the changes with the address. Am I wrong?

dimerr, 2023-01-15:
Yeah, really weird, hal_hisi is missing

Moleus, 2023-01-15:
It works!
```
sensors:
- vendor: GalaxyCore
  model: GC1024
  control:
    bus: 0
    type: i2c
    addr: 0x78
  data:
    type: DC
  clock: 24MHz
```
Thank you!

closed 2023-01-15 by Moleus