OpenIPC/ipctool #53: Not support this sensor

vinibali, 2023-02-12:


dimerr, 2023-02-12:
so which sensor you actually have there?

vinibali, 2023-02-12:
I don't really know. It's a 5MP one.
* the panel with IR leds looks like this (FY-S63): [https://www.tradechina.com/FY-S63-3-smd-led-ir-Array-pcb-board-CCTV-Accessories_192493008.html](https://www.tradechina.com/FY-S63-3-smd-led-ir-Array-pcb-board-CCTV-Accessories_192493008.html)
* the camera itself have the following numbers: cmos0253p-38x38-wp1 and f20201222v

vinibali, 2023-02-12:
I think I found it:
* 0235p-5333p: (5.0M Black light illumination H.265 AI IP Camera Module)[https://www.xiongmaitech.com/en/index.php/product/product-detail/185/187/377]

vinibali, 2023-02-14:
https://github.com/OpenIPC/firmware/issues/64

vinibali, 2023-02-18:
Is there any way, how I can help to get his supported?

widgetii, 2023-02-18:
You should make changes in source code, test on your hardware and make a PR

vinibali, 2023-02-18:
Can you please help meg and make a bit more detailed guide?

widgetii, 2023-02-18:
Do you have C programming skills?

vinibali, 2023-02-19:
sort of

widgetii, 2023-02-19:
Then do debug ipctool binary on your hardware figuring out why sensor is not responding to i2c commands

vinibali, 2023-02-19:
it seems, i've attached the cable to the sensor wrongly :)
```
root@openipc-xm530:/tmp# ipctool 
---
chip:
  vendor: Xiongmai
  model: XM550
board:
  vendor: Xiongmai
ethernet:
  mac: "xx:xx:xx:xx:xx:xx"
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 9d2158df
        contains:
          - name: xmcrypto
            offset: 0x2fc00
          - name: uboot-env
            offset: 0x30000
      - name: env
        size: 0x10000
        sha1: b0152796
      - name: kernel
        size: 0x200000
        sha1: 86f4024b
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: 55c76271
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
ram:
  total: 128M
  media: 93M
firmware:
  kernel: "3.10.103+ (SMP Sat Feb 11 23:00:38 UTC 2023)"
  toolchain: gcc version 8.4.0 (Buildroot -gff5aa3b6-dirty) 
  main-app: /usr/bin/majestic
sensors:
- vendor: SmartSens
  model: SC5332
  control:
    bus: 0
    type: i2c
    addr: 0x60
```

vinibali, 2023-02-19:
I saw a couple of references for this sensor string in the SDK.

vinibali, 2023-02-25:
Maybe, I didn't explained the finding. So the camera module was not found because the cable was attached wrongly.
However there is still no picture in majestic. Back in last time, I compiled the ipctool, and it seems there are no other i2s devices attached.
```
root@openipc-xm530:/tmp# ./ipctool i2cdetect
       0  1  2  3  4  5  6  7   8  9  a  b  c  d  e  f
    : xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  10: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  20: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  30: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  40: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  50: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  60: 60 61 xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  70: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  80: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  90: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  a0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  b0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  c0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  d0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  e0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx xx  |  
  f0: xx xx xx xx xx xx xx xx  xx xx xx xx xx xx xx
```

Dmesg still shows some i2c timeout for address 0x2
```
<3>i2c_write timeout 0x2
<3>i2c_write timeout 0x2
```
Majectic's output hasn't changed since.
Is that possible, that this sensor is supported, and just the build is a little bit older?

dimerr, 2023-02-25:
Nevertheless this sensor is not supported, where have you seen it in sdk?

vinibali, 2023-02-26:
I didn't took a minute to understand the SDK, however this string can be found on many places:
[https://github.com/OpenIPC/openxiongmai/search?q=SC5332](https://github.com/OpenIPC/openxiongmai/search?q=SC5332)

dimerr, 2023-02-26:
In that case it should work, there was some symlink to libsns in /usr/lib , check if it points to X50 lib

vinibali, 2023-02-26:
That was possibly the problem, the symlink was pointing to the libsns_X123_XM530.so.
I changed it to libsns_X50_XM530.so in the filesystem and in the initfile too, but that broke something.
The device falled into a reboot loop, no matter what I was doing, however I had around 5 secs on each run to remove the symlink and the new case in the start function in the initfile.
I was following the logread and these was the latest lines:
```
Feb 26 18:42:06 openipc-xm530 kern.err kernel: i2c_write timeout 0x2
Feb 26 18:42:06 openipc-xm530 kern.emerg kernel: IspKo: [IspDrv]: Memery Init Buffer!Num[2]
client_loop: send disconnect: Broken pipe
```
According to a not working, but not rebooting state, majestic caused some problem with it's start.
```
Feb 26 18:43:05 openipc-xm530 kern.err kernel: i2c_write timeout 0x2
Feb 26 18:43:05 openipc-xm530 kern.err kernel: i2c_write timeout 0x2
Feb 26 18:43:05 openipc-xm530 user.info majestic[621]: ^M
Feb 26 18:43:05 openipc-xm530 kern.emerg kernel: IspKo: [IspDrv]: Memery Init Buffer!Num[2]
Feb 26 18:43:05 openipc-xm530 kern.emerg kernel: mve_rsrc_data.hw_version=56550002
Feb 26 18:43:07 openipc-xm530 user.err majestic[621]: Cannot create VENC chn 0
Feb 26 18:43:07 openipc-xm530 user.err majestic[621]: XM_ERR_VENC_NOT_CONFIG: try to enable or initialize system, device or channel, before configuring an attribute
Feb 26 18:43:07 openipc-xm530 user.err majestic[621]: Start VencChn 0 failed!
Feb 26 18:43:07 openipc-xm530 user.info majestic[621]: HTTP server started on 0.0.0.0:80
Feb 26 18:43:07 openipc-xm530 user.info majestic[621]: RTSP server started on port 554
Feb 26 18:43:07 openipc-xm530 user.info majestic[621]: NETIP server started on port 34567
Feb 26 18:43:13 openipc-xm530 cron.info crond[480]: time disparity of 21335 minutes detected
```
This happens, when I run majestic with the updated symlink:
```
root@openipc-xm530:~# /usr/bin/majestic 
LIBH264: Complied at Apr 30 2019 16:55:22
18:52:21  <       majestic> [    main] main@125                      Majestic Lite for Xiongmai, version HEAD+eba3d1e, built on 2023-02-11
18:52:21  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
18:52:21  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
DEBUG: [ISP_SDK]BUILD TIME:Sep 21 2021 11:58:04
DEBUG: OSC:12M
18:52:21  <       majestic> [     log] printf@268                    SYS/VI set in	[ Sofia ]
18:52:21  <       majestic> [     log] printf@268                    Vstd_init	[ PAL ]
18:52:21  <       majestic> [     log] printf@268                    Rslt	[ 0 ]
18:52:21  <       majestic> [     log] printf@268                    IPC_Venc:	[ H.264 ]
18:52:21  <       majestic> [     log] printf@268                    Sns_IF:	[ By src ]

ERR: sensor_set_chip not support this sensor!
ERR: open /mnt/mtd/Config/SensorType.bat failed
ERR: sensor_set_chip not support this sensor!
DEBUG: DspChip:XM530
DEBUG: SnsIF:DVP
DEBUG: Vstd:PAL[25fps]
ERR: [sensor_init]this sensor chip unsupport!
Segmentation fault
```

dimerr, 2023-02-26:
What init file? As i remember, only symlink has to be switched and theres even a script that does that https://github.com/OpenIPC/firmware/blob/master/general/package/xiongmai-osdrv-xm530/files/script/S95xiongmai#L45 but that happens only once, if symlink isnt there, but your sensor doesnt fall under condition, thats why it was pointing at X123

Check /proc/cmdline , mem and osmem should be same and equal to 35M on X123 and prolly should be 64M on X50

vinibali, 2023-02-27:
As the initfile, I was talking about the S95xiongmai, I understood how it works.
In case of the osmem, that's an env variable at U-boot. I've increased that value to 64M and the mem parameter too, which is in the bootarg. But the outcome is pretty much the same.
```
root@openipc-xm530:~# fw_printenv | grep osmem
bootargs=mem=64M osmem=64M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=xm_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
osmem=64M
```
I've tried to run majestic with the ipctool's strace, but it doesn't show much more information.
```
root@openipc-xm530:/tmp# ./ipctool trace /usr/bin/majestic 

[679] child 680 created
LIBH264: Complied at Apr 30 2019 16:55:22
10:53:09  <       majestic> [    main] main@125                      Majestic Lite for Xiongmai, version HEAD+eba3d1e, built on 2023-02-11
10:53:09  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
10:53:09  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
Cloned 6 fds

parent 680 created child 681
Cloned 6 fds

parent 680 created child 682
Cloned 9 fds

parent 680 created child 683
DEBUG: [ISP_SDK]BUILD TIME:Sep 21 2021 11:58:04
DEBUG: OSC:12M
10:53:10  <       majestic> [     log] printf@268                    SYS/VI set in      [ Sofia ]
10:53:10  <       majestic> [     log] printf@268                    Vstd_init  [ PAL ]
10:53:10  <       majestic> [     log] printf@268                    Rslt       [ 0 ]
10:53:10  <       majestic> [     log] printf@268                    IPC_Venc:  [ H.264 ]
10:53:10  <       majestic> [     log] printf@268                    Sns_IF:    [ By src ]

ERR: sensor_set_chip not support this sensor!
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
ERR: open /mnt/mtd/Config/SensorType.bat failed
ERR: sensor_set_chip not support this sensor!
DEBUG: DspChip:XM530
DEBUG: SnsIF:DVP
DEBUG: Vstd:PAL[25fps]
ERR: [sensor_init]this sensor chip unsupport!
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
usleep(100000)
[REBOOT]
```

vinibali, 2023-02-27:
Hmm, that's really strange. The lib doesn't contain that string.
```
# strings /usr/lib/sensors/libsns_X50_XM530.so   | grep -i sc5
cmos_init_sensor_exp_function_sc5235
cmos_init_ae_exp_function_sc5235
cmos_init_awb_exp_function_sc5235
sensor_init_sc5235
cmos_init_sensor_exp_function_SC5239
cmos_init_ae_exp_function_SC5239
cmos_init_awb_exp_function_SC5239
sensor_init_sc5239
[0;32;32mDEBUG: ------------- SC5300 5M 15fps  init ok! (@201801225_158x4)----------------
[0;32;32mDEBUG: ------------- SC5235 5M 15fps  init ok! (@20180312_158x4)----------------
[0;32;32mDEBUG: ------------- SC5239 4M 15fps  init ok! (@201801225_158x4)----------------
[0;32;32mDEBUG: SC5239!
```

vinibali, 2023-03-02:
Can you please give a wiki page or documentation, which helps me to rebuild the firmware?

themactep, 2023-03-02:
https://github.com/OpenIPC/wiki/blob/master/en/bulding.md

vinibali, 2023-03-18:
It seems the existing libraries are old and doesn't have the support for SC5332 camera module.
I recompiled libsns_X50_XM530.so, but the case is pretty much the same.
```
root@openipc-xm530:/usr/lib# strings sensors/libsns_X50_XM530.so | grep -i sc5332
cmos_init_sensor_exp_function_sc5332
cmos_init_ae_exp_function_sc5332
cmos_init_awb_exp_function_sc5332
sensor_init_sc5332
[0;32;32mDEBUG: ------------- sc5332 5M 15fps  init ok! (@20200226_158x4)----------------
root@openipc-xm530:/usr/lib# /usr/bin/majestic
LIBH264: Complied at Apr 30 2019 16:55:22
06:28:09  <       majestic> [    main] main@125                      Majestic Lite for Xiongmai, version HEAD+eba3d1e, built on 2023-02-11
06:28:09  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
06:28:09  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
DEBUG: [ISP_SDK]BUILD TIME:Sep 21 2021 11:58:04
DEBUG: OSC:12M
06:28:09  <       majestic> [     log] printf@268                    SYS/VI set in	[ Sofia ]
06:28:09  <       majestic> [     log] printf@268                    Vstd_init	[ PAL ]
06:28:09  <       majestic> [     log] printf@268                    Rslt	[ 0 ]
06:28:09  <       majestic> [     log] printf@268                    IPC_Venc:	[ H.264 ]
06:28:09  <       majestic> [     log] printf@268                    Sns_IF:	[ By src ]

1sensor_set_chip not support this sensor 0!
DEBUG: SENSOR ID = ce1a
ERR: open /mnt/mtd/Config/SensorType.bat failed
1sensor_set_chip not support this sensor 130!
DEBUG: DspChip:XM530
DEBUG: SnsIF:DVP
DEBUG: Vstd:PAL[25fps]
[sensor_init]this sensor chip unsupport 130!
Segmentation fault
```
Is it possible that other libraries needs to be recompiled again?
I even made a PR to get the repo aligned.
https://github.com/OpenIPC/openxiongmai/pull/1