OpenIPC/ipctool #77: Could not start ipctool on original firmware

a-baturin, 2023-06-30:
Hi and thanks for your work.

I have Hi3518ev100 (ESCAM QF001) camera.
I could not run ipctool on *original* firmware. I could install OpenIPC and ipctool works there, but I think sensor detected incorrectly.
```
chip:
  vendor: HiSilicon
  model: 3518EV100
board:
  vendor: OpenIPC
  version: 2.3.06.14
ethernet:
  mac: "00:12:16:b4:3c:3e"
  u-mdio-phyaddr: 1
  phy-id: 0x02430c54
  d-mdio-phyaddr: 0
  phy-mode: mii
rom:
  - type: nor
    block: 64K
    partitions:
      - name: boot
        size: 0x40000
        sha1: 42f2bad9
      - name: env
        size: 0x10000
        sha1: b82436a1
        contains:
          - name: uboot-env
            offset: 0x0
      - name: kernel
        size: 0x200000
        sha1: 8259e82a
      - name: rootfs
        size: 0x500000
        path: /,squashfs
        sha1: e9ba9523
      - name: rootfs_data
        size: 0xb0000
        path: /overlay,jffs2,rw
    size: 8M
    addr-mode: 3-byte
ram:
  total: 64M
  media: 32M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:25:39)"
  kernel: "3.0.8 (Wed Jun 14 10:11:57 UTC 2023)"
  toolchain: buildroot-gcc-12.2.0
  sdk: "Hi3518_MPP_V1.0.B.0  (Nov 19 2015, 16:37:04)"
sensors:
- vendor: Silicon Optronics
  model: JXH42
  control:
    bus: 0
    type: i2c
    addr: 0x60
  vicap-state: down
```
So in original FW I got:
```
hisilicon # tftp 0x82000000 ipctool
Hisilicon ETH net controler
miiphy_register: non unique device name '0:1'
miiphy_register: non unique device name '0:2'
MAC:   00-12-16-B4-3C-3E
UP_PORT : phy status change : LINK=UP : DUPLEX=FULL : SPEED=100M
TFTP from server 192.168.1.86; our IP address is 192.168.1.38
Download Filename 'ipctool'.
Download to address: 0x82000000
Downloading: #################################################
done
Bytes transferred = 128244 (1f4f4 hex)
hisilicon # go
## Starting application at 0x82000000 ...
undefined instruction
pc : [<82000008>]          lr : [<80807804>]
sp : 807cf1f0  ip : 0000001c     fp : 807cf629
r10: 807cfa29  r9 : 000003f3     r8 : 807cffe0
r7 : 00000002  r6 : 82000000     r5 : 807cfe20  r4 : 00000002
r3 : 82000000  r2 : 807cfe20     r1 : 807cfe20  r0 : 00000001
Flags: nZCv  IRQs off  FIQs off  Mode SVC_32
Resetting CPU ...

resetting ...

```

Am I doing something wrong?

themactep, 2023-06-30:
You need ro run ipctool in stock linux, not in bootloader shell.

a-baturin, 2023-06-30:
Oh, that's sad but explains the issue.
Unfortunately I could not reach a stock linux shell yet, will try harder.

closed 2023-06-30 by a-baturin