OpenIPC/ipctool #103: gpio scan issue

bedenko, 2023-12-10:
I have module IMA20S07 with SSC335 and IMX307.
I've flashed module with latest image 2.3.12.07-lite.
IRCUT is not working, so i ran command `ipctool gpio scan`, but i got empty return with exit code 1.

I tried `ipctool trace /tmp/ipctool gpio scan`, but ipctool creates child process and trace tool does not support tracing of child processes.

Is there any other way of debugging this? Besides setting up complete SDK and compiling new ipctool with additional outputs.

widgetii, 2023-12-10:
`Besides setting up complete SDK and compiling new ipctool with additional outputs.`

That's the only way to debug it properly

FlyRouter, 2023-12-11:
One simple and dirty way is to try to control different GPIOs by brute force in scanning mode and changing values.
The last two numbers indicate from which GPIO you need to scan. 
Break it up into small sections and keep an eye on the camera. 
Quite often, an IR filter is attached to 78 and 79 GPIOs, but this is only one of the options.

```
curl -s https://raw.githubusercontent.com/OpenIPC/sandbox/main/scripts/search-gpio/search-gpio | sh -s -- 0 100
```

FlyRouter, 2023-12-12:
Good
Don’t forget to then indicate the data for your device in the GPIO table in our Wiki.

closed 2024-03-18 by dimerr