OpenIPC/ipctool #112: Same Chip ID on 3 sensors are not distinguished

RoboSchmied, 2024-02-29:
There are **3 sensors** in the openipc repos that have the **same chip ID  `cb3e`** but are all seen as `sc223a`.
- [sc223a](https://github.com/OpenIPC/openingenic/blob/3a9fdfb35c6e45f15d3aaebaf9ca8bd77914c7a1/kernel/sensors/t31/sc223a/sc223a.c#L26) ([30 fps](https://smartsens.oss-cn-beijing.aliyuncs.com/web/products/SC223A_V3.0.pdf))
- [sc233a](https://github.com/OpenIPC/openingenic/blob/3a9fdfb35c6e45f15d3aaebaf9ca8bd77914c7a1/kernel/sensors/t41/sc233a/sc233a.c#L28) ([60 fps](https://smartsens.oss-cn-beijing.aliyuncs.com/web/products/SC233A_V3.0.pdf)) 
- [sc2239p](https://github.com/OpenIPC/openingenic/blob/3a9fdfb35c6e45f15d3aaebaf9ca8bd77914c7a1/kernel/sensors/t31/sc2239p/sc2239p.c#L26)

[ipctool handling `cb3e`](https://github.com/OpenIPC/ipctool/blob/3e7944bea7b8fe0956cc1f5159cbac33ba4ea850/src/sensors.c#L593)

[Here](https://github.com/OpenIPC/firmware/issues/1324) we found out that there may be different "versions" of the sc223a.

Thanks to @johndi3 who [mentioned](https://github.com/OpenIPC/firmware/issues/1324#issuecomment-1970326960) sc233a and thanks to @themactep who helped [discovering the ChipID](https://github.com/OpenIPC/firmware/issues/1324#issuecomment-1969336663).

As shown [here](https://github.com/OpenIPC/ipctool/blob/3e7944bea7b8fe0956cc1f5159cbac33ba4ea850/src/sensors.c#L493) we should regard more registers to differentiate between the 3 sensors.

**If you have one of the 3 sensors, please post the result of this command:** 
```
ipctool i2cdump 0x60 0x3100 0x310F`.
```
That will show your device ID + some more registers.

The output of my "version" of the sc223a (sold as sc5239s) is:
```
root@openipc-gk7205v210:~# ipctool i2cdump 0x60 0x3100 0x310F
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
3100: 00 12 00 08 01 12 05 CB  3E 01 00 00 00 00 00
```
The important part here is `CB  3E 01`.   Thank you.

widgetii, 2024-03-01:
I'd start from finding sensor's datasheet

themactep, 2024-03-01:
Hard does not mean impossible. I have a few here https://github.com/themactep/openipc-wiki/tree/master/hardware/components/sensor/docs/ but not for these particular models.

RoboSchmied, 2024-03-01:
> Hard does not mean impossible. I have a few here https://github.com/themactep/openipc-wiki/tree/master/hardware/components/sensor/docs/ but not for these particular models.

Wow, thank you. Very interesting. How + where did you get them?

themactep, 2024-03-01:
I know a guy who knows a guy, you know... LOL. Google and other search engines can do magic.

RoboSchmied, 2024-03-02:
They are marked as `Company Confidential` and `SmartSens Confidential - NDA only`. 
No chance for me with google & co.

closed 2026-09-16 by openipc-ai