OpenIPC/ipctool #126: ipctool segfaults on xm510

nickoe, 2024-04-03:
Hi

I am playing around with a camera I have laying around, I think we identify this as an xm510. I got telnet access with the Snawoot/hisilicon-dvr-telnet tool.

I tried to download the ipctool binary on https://github.com/OpenIPC/ipctool/releases/download/latest/ipctool which appears to have been updated two days ago.

```
$ sha256sum ipctool 
8d0f7c48c1f6a616d39bff8dc41c1e2ac86b1b6276995eca0211eecd29cd97eb  ipctool
$ file ipctool 
ipctool: ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), statically linked, no section header
```
My device reports:
```
~ # cat /proc/cpuinfo
cat /proc/cpuinfo
Processor       : ARMv7 Processor rev 1 (v7l)
BogoMIPS        : 199.06
Features        : swp half fastmult edsp 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc05
CPU revision    : 1

Hardware        : xm510
Revision        : 0000
Serial          : 0000000000000000
cat /proc/kmsg
<5>Linux version 3.0.101 (jinze@xd-server-0001) (gcc version 4.9.2 (Buildroot 2014.08) ) #11 Tue Jun 20 08:36:44 CST 2017
<4>CPU: ARMv7 Processor [410fc051] revision 1 (ARMv7), cr=10c53c7d
...
```

So I guess that it segfaults because it is not built for the correct abi? Does a version for ARMv7 exist somewhere precomipled?

nickoe, 2024-04-03:
I even tried to do:
```
$ wget https://musl.cc/armv7l-linux-musleabihf-cross.tgz
$ tar xvf armv7l-linux-musleabihf-cross.tgz
cmake -DCMAKE_C_COMPILER=$(pwd)/armv7l-linux-musleabihf-cross/bin/armv7l-linux-musleabihf-gcc -DCMAKE_BUILD_TYPE=Release
make
$ file ipctool 
ipctool: ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), statically linked, no section header
```

But it appears to result in the same elf file format and it also segfaults.

sansarus, 2025-01-01:
maybe you have dir names with "." at /proc/.